OPCFoundation/UA-EdgeTranslator #6: Question: IEC 61850 southbound interface

JonasDaenzer, 2023-11-09:
Do you have an IEC 61850 southbound interface planned?

barnstee, 2023-11-09:
No, but feel free to contribute one. ;-)

barnstee, 2024-08-16:
Reopening as we now have support for Siemens, Beckhoff, Rockwell, Mitsubishi, OPC UA and Modbus. @JonasDaenzer, it is literally 1 day of work to add support for a new low-level PLC protocol as we managed to further simplify the IAsset interface. Please have a look at the existing implementations and consider a contribution for IEC 61850.

JonasDaenzer, 2024-08-16:
I am not a programmer, I have no idea how to do. Sorry

barnstee, 2025-03-19:
Experimental implementation now available with https://github.com/OPCFoundation/UA-EdgeTranslator/commit/f86dade21ee84460a7357f0b407dc4b099a81cd1.

closed 2025-03-19 by barnstee