OPCFoundation/UA-EdgeTranslator #28: Example Modbus TD file fails JSON-LD validation

rlabbeptc, 2024-03-27:
Looks like the example TD fails validation against the latest MODBS binding Schema [MODBUS Binding Schema](https://w3c.github.io/wot-binding-templates/bindings/protocols/modbus/modbus.schema.json). I was using the [python-jsonschema](https://python-jsonschema.readthedocs.io/en/stable/#) package to validate.

[Pac4200 Example TD](https://github.com/OPCFoundation/UA-EdgeTranslator/blob/main/Samples/pac4200.td.jsonld)

The example's issue is the Modbus register types casing doesn't match the schema's definitions. I wasn't sure if this is a schema issue or if the example needs to be update. 

If it's an issue with the Pac4200 Example I can do a pull request to fix it. I just wanted to confirm before creating one. Just let me know.

egekorkan, 2024-03-27:
It seems to be a minor issue in the provided TD. `holdingRegister` in the forms should be `HoldingRegister`

barnstee, 2024-03-28:
Fixed with https://github.com/OPCFoundation/UA-EdgeTranslator/commit/ee8e1169ef10ccb559169a7812d1f2def344785a.

closed 2024-03-28 by barnstee