OPCFoundation/UA-EdgeTranslator #29: Add Variable node to return the list of WoT protocol bindings a UA server supports

barnstee, 2024-04-16:
This can be implemented as a simple data node returning an enumeration of the supported WoT protocol bindings:
Released: https://www.w3.org/TR/wot-binding-templates/#table-existing-protocol-binding-templates
All bindings: https://w3c.github.io/wot-binding-templates/#protocol-bindings-table

jestun, 2024-04-16:
I would say that there should be something like a "WoTBindingProtocolEnum" enumeration DataType listing all possible binding protocols (the actual content/info probably need to be discussed).
Then there could be an optional variable node like "SupportedBindingProtocols" in the WoTAssetConnectionManagementType that would provide an array including enum members of the supported binding protocols.

barnstee, 2024-06-11:
Implement as UA profile type and return list of profiles.

barnstee, 2024-06-25:
This has been added to the spec, v1.0.1

barnstee, 2024-07-14:
This is added to this reference implementation with PR https://github.com/OPCFoundation/UA-EdgeTranslator/commit/c5d2169e9d3c8c5be5a7e050a1b69c95781aa22f

barnstee, 2024-07-23:
Closing as feature complete.

barnstee, 2024-07-23:
Mantis issue for tracking spec change: https://mantis.opcfoundation.org/view.php?id=9679