OPCFoundation/UA-EdgeTranslator #39: CAN DBC to WoT

dwhiteddsoft, 2024-12-13:
Has anyone thought of converting from a CAN DBC file to WoT and then using this against a device that speaks CAN?

barnstee, 2024-12-13:
I don't think so.

sebastiankb, 2024-12-13:
does CAN DBC contain address information or only data CAN-based data values?

dwhiteddsoft, 2024-12-13:
The DBC file is CAN IDs and data. https://www.csselectronics.com/pages/can-dbc-file-database-intro

I am very ignorant aobut this source so please forgive me but are the addresses necessary? Most times you simply get on the network and start decoding messages. I have always thorught of it in a broadcast mode :-)

dwhiteddsoft, 2024-12-13:
Just so everyone knows, we are looking at creating a CAN -> OPC UA protocol/message translator. This project seems to do a lot of what we need and I already have a whole library that does CAN decoding. Just asking questions before I decide whether this project is the right place to extend and leverage the work here or simply build a bespoke translator

barnstee, 2024-12-13:
Extend this project, of course! :-)