OPCFoundation/UA-EdgeTranslator #42: Mapping of WoT Alarms

barnstee, 2025-05-15:


barnstee, 2025-07-22:
https://www.w3.org/TR/wot-thing-description11/#event-serialization-json

sebastiankb, 2025-07-24:
we could already use the `uav:mapToType` from the WoT UA Binding to annotate the event to which ConditionType it is mappable in OPC UA. E.g,

```json
{
  {"@context": ["https://www.w3.org/2022/wot/td/v1.1",
    {
      "0": "http://opcfoundation.org/UA/",
      "uav": "http://opcfoundation.org/UA/WoT-Binding/"
    }],
    // ...
    "events": {
        "overheated": {
            "uav:mapToType" : "0:LimitAlarmType",
           "data": {
                "type": "string"
            },
            "forms": [...]
        }
    },
    // ...
}
```

In the case there is no specific alarm type provided in the TD, a mapping to a simple UA Variable can be considered as default.

barnstee, 2026-09-23:
Now a duplicate of https://github.com/OPCF-Members/OPC10100-WoT/issues/10

closed 2026-09-23 by barnstee