OPCFoundation/UA-EdgeTranslator #48: Rockwell client not writing to PLC

jdieringer, 2025-08-29:
I observed that the Rockwell Client was not writing the value to the PLC when writing to the variable in OPC-UA. The value would change momentarily but then revert back to the PLC value on next scan. In addition, the value never changed in the PLC. 

I added a tag.Write(); to the Write task in RockwellClient.cs which seems to resolve this. I guess the tag.SetXXX method is only setting the desired value locally but a tag.Write() is needed to send the write to the PLC. 

```
            tag.Read();

            int offset = unitID;

            switch (function)
            {
                case "BOOL": tag.SetBit(offset, BitConverter.ToBoolean(values)); break;
                case "SINT": tag.SetInt8(offset, (sbyte) BitConverter.ToChar(values)); break;
                case "INT": tag.SetInt16(offset, BitConverter.ToInt16(values)); break;
                case "DINT": tag.SetInt32(offset, BitConverter.ToInt32(values)); break;
                case "LINT": tag.SetInt64(offset, BitConverter.ToInt64(values)); break;
                case "USINT": tag.SetUInt8(offset, values[0]); break;
                case "UINT": tag.SetUInt16(offset, BitConverter.ToUInt16(values)); break;
                case "UDINT": tag.SetUInt32(offset, BitConverter.ToUInt32(values)); break;
                case "ULINT": tag.SetUInt64(offset, BitConverter.ToUInt64(values)); break;
                case "REAL": tag.SetFloat32(offset, BitConverter.ToSingle(values)); break;
                case "LREAL": tag.SetFloat64(offset, BitConverter.ToDouble(values)); break;
                default: break;
            }
            //New code
            tag.Write();
```

barnstee, 2025-08-29:
I think I figured this out: https://github.com/OPCFoundation/UA-EdgeTranslator/commit/5ea367d4a8c69f6e6f4ed59bf207be8542f25d79. Correct?

jdieringer, 2025-09-08:
Apologies, notifications went to the wrong inbox. Yes, looks great!

closed 2025-09-08 by jdieringer