OPCFoundation/UA-EdgeTranslator #49: how to use it

daodol, 2025-09-05:
how to use it

daodol, 2025-09-09:
@barnstee I have already installed it through Docker and can access port 19521. What should I do next? I am not clear about it now.

barnstee, 2025-09-09:
From the port you want to access, I can tell you are interested in OCPP support. Are you familiar with OCPP and OPC UA? UA Edge Translator acts as an OCPP Central System and as an OPC UA server in this scenario. I need a lot more information on what you are trying to achieve to be able to help you.

daodol, 2025-09-09:
@barnstee I am familiar with OPCUA and Modbus.I also know the functions of this project, it can “translate” modbus-tcp into opcua protocol.But how should I operate to achieve such an effect?

barnstee, 2025-09-09:
> [@barnstee](https://github.com/barnstee) ok,Do you mean it will provide the relevant endpoint through OPC UA? By calling the endpoint, the TD file will be passed to it, thereby enabling the “translation” function?
> 
> [image]

Correct. See the samples directory for sample TDs, for example a Modbus one: https://github.com/OPCFoundation/UA-EdgeTranslator/blob/main/Samples/pac4200.td.jsonld

daodol, 2025-09-10:
@barnstee I have a question. It can convert a protocol into OPC UA, so the TD file should describe the original protocol address as well as the corresponding OPC UA protocol address (node). However, from the example file, I only see descriptions related to the OPC UA protocol address (node), and I do not see descriptions of the original protocol address.

[image]

barnstee, 2025-09-10:
The forms section is always related to the original protocol address, specifically the href. It's a bit hard to read, but in your example the original address is Modbus holding register address 1 and the variable is 2 Modbus registers wide (quantity), i.e. 32 bits (as since it's a float). If you search for "W3C WoT Modbus protocol binding" on the web, you will find the spec explaining all this in detail.